nerdyscout/Arduino_MCP3x6x_Library #1: Problems with SDO on MCP3464

Omikron90, 2022-01-19:
Hi, 
first of all thank you very much for the code.  I was very happy to find something for this ADC family. 
Unfortunately the code does not run on my MCP3464 yet. The signals from the CLK, CS and SDI look good but unfortunately the ADC at the SDO gives nothing out. Only a small ripple from the CLK is visible on the Scope.  Do you have an idea where the problem could be?

I would be very grateful for any information.
Thanks a lot

nerdyscout, 2022-01-19:
hey @Omikron90 
please be aware this is a very early stage of a library, and not functional at all!

I do not own a MCP346x, but according the datasheet it should behave similar to my MCP3564.

Your description sound familiar to me...
```
**chapter 6.2.4**
Changing the command type within the same communication (while CS is
logic low) is not possible. The communication has to be
stopped (CS rising edge) and restarted (CS falling
edge) to change its command type. The list of possible
commands, their type and their possible command
addresses are described in Table 6-2
```
maybe check if CS toggles on every command written, that solved it for me.

Omikron90, 2022-01-25:
hey @nerdyscout 
thanks for the two ideas, especially the pullup on MCLK has brought me further. Now I can write and read the registers :-) 

Yes, the MCP346x is for the most part the same as the MCP356x. I should have considered however actually also all changes when writing the registers. 

Unfortunately the ADC still does not return any values when I read the register 0x0 (ADCDATA). Did you also have problems with this?

nerdyscout, 2022-01-25:
default it expects a external clk of 4.9152Mhz used for ADC sampling. but you could configure an internal clk. that should help.

closed 2022-02-04 by nerdyscout