nerdyscout/Arduino_MCP3x6x_Library #3: Not really an issue :)

mnemocron, 2022-08-19:
Last year you discovered my repo for this chip and even helped out.
https://github.com/fhnw-ise-qcrypt/mcp3564

I just wanted to let you know that my library is now at a point where it is working in a product that I am still developing.
https://github.com/fhnw-ise-qcrypt/stm32-wiggly-pointy

And I came up with a conf.h file to have various configuration examples because this chip is just so damn adaptable.
So feel free to cheat by looking through my code. This is how I handle the "streaming" mode via GPIO interrupts. It actually works according to the timing from the Excel sheet:
https://github.com/fhnw-ise-qcrypt/stm32-wiggly-pointy/blob/main/Core/Src/stm32f3xx_it.c#L225

I am really curious how you implement the `Stream` class as I have seen in the example that you want to achieve.

nerdyscout, 2022-08-20:
hi @mnemocron! great to hear you are doing progress... I am currently a bit stuck with this lib. So, for sure, I will have a look at your code.

Actually I do not really know what to do with the Stream interface. my basic idea was to put the ADC in scan mode with static conversion and setup some interrupt handler to enable dma based spi transfer. afaik this is mostly what you are doing as well... but I still see no reason why to use the stream interface... might have been just a stupid idea when I initially set this up.

mnemocron, 2022-08-30:
>  my basic idea was to put the ADC in scan mode with static conversion and setup some interrupt handler to enable dma based spi transfer.

Yeah, that is what I had in mind first. But I was not able to use DMA since I suppose this is very much dependent of the hardware implementation of the SPI peripheral.
In streaming mode, the ADC is the master that generates DRDY interrupts for each sample. 
The MCU then has to react on that IRQ signal on a GPIO. Unless you can configure a DMA transfer of 3-4 bytes each time a trigger is present, i don't think DMA works. 
But good luck anyhow. I saw you are using the ESP32.

nerdyscout, 2022-08-31:
[image]
well with my latest approch (code not yet published) I do get an IRQ every 670µs (yellow), which triggers an interrupt handler reading 4 bytes wich lasts 11µs. strictly technical speaking this might be not DMA, but it is still the quickest i can think of.

closed 2022-09-04 by nerdyscout