nerdyscout/Arduino_MCP3x6x_Library #4: Get stuck in mcp.analogRead loop

FelixWT, 2022-12-09:
Hi, 
Thank you for making this ADC library. It's one of the few library that support MCP3561. 

But when I'm implement this code to my ESP32 S3 it seems to stuck in analogread infinite loop.
The differences are I make the  MCP3561 instance aka `mcp` a global instance by adding `extern MCP3561 mcp;` in MCP3x6x.h and `MCP3561 mcp=MCP3561();` in MCP3x6x.cpp . 
Also dude to the layout error in swap the pin definition of MISO & MOSI in the class Constructor .

```
MCP3x6x::MCP3x6x(const uint16_t MCP3x6x_DEVICE_TYPE, const uint8_t pinCS, SPIClass *theSPI,
                 const uint8_t pinMOSI, const uint8_t pinMISO, const uint8_t pinCLK) {
  switch (MCP3x6x_DEVICE_TYPE) {
    case MCP3461_DEVICE_TYPE:
      _resolution_max = 16;
      _channels_max   = 2;
      break;
    case MCP3462_DEVICE_TYPE:
      _resolution_max = 16;
      _channels_max   = 4;
      break;
    case MCP3464_DEVICE_TYPE:
      _resolution_max = 16;
      _channels_max   = 8;
      break;
    case MCP3561_DEVICE_TYPE:
      _resolution_max = 24;
      _channels_max   = 2;
      break;
    case MCP3562_DEVICE_TYPE:
      _resolution_max = 24;
      _channels_max   = 4;
      break;
    case MCP3564_DEVICE_TYPE:
      _resolution_max = 24;
      _channels_max   = 8;
      break;
    default:
#warning "undefined MCP3x6x_DEVICE_TYPE"
      break;
  }

  //  settings.id = MCP3x6x_DEVICE_TYPE;
/* ========================================= */

  _spi        = theSPI;
  _pinMISO    = pinMOSI;//SWAP DUE TO LAYOUT ERROR
  _pinMOSI    = pinMISO;//SWAP DUE TO LAYOUT ERROR
  _pinCLK     = pinCLK;
  _pinCS      = pinCS;

  Serial.print("_pinMISO: ");
  Serial.println(_pinMISO);
  Serial.print("_pinMOSI: ");
  Serial.println(_pinMOSI);
  Serial.print("_pinCLK: ");
  Serial.println(_pinCLK);
  Serial.print("_pinCS: ");
  Serial.println(_pinCS);

/* ========================================= */

  _resolution = _resolution_max;
  _channel_mask |= 0xff << _channels_max;  // todo use this one
};
```
And this is my main task
```
#include <SPI.h>
#include "MCP3x6x.h"

void setup()
{  
  Serial.begin(115200);  
  MCP3561 mcp(10); 
  while (!mcp.begin()) {
    Serial.println("failed to initialize MCP");
    delay(200);
  }
}
void loop()
{
  Serial.println("1");//stuck detect
  int32_t adcdata = mcp.analogRead(MCP_CH0);
  Serial.println("2");//stuck detect
  double voltage = adcdata * mcp.getReference() / mcp.getMaxValue();  
  Serial.print("voltage: ");
  Serial.println(voltage, 10);
  delay(1000);
}
```
And the Serial out put:
_pinMISO: 11
_pinMOSI: 13
_pinCLK: 12
_pinCS: 10
1
mux

Any advice is greatly appreciated.

Mirageofmage, 2022-12-09:
Hey Felix,

I've run into this exact issue (also with the ESP32S3). I'm currently working on a PR to make this library fully compatible, but for the time being here's the fix in order to get analogRead to work.

`In MCP3x6x.h`
```
typedef union {
     struct {
       struct __attribute__((__packed__)) {
         bool por;     //!< status: power on reset
         bool crccfg;  //!< status: crc
         bool dr;      //!< status: data ready
       };
       uint8_t      : 1;  //!< !addr[0]
       uint8_t addr : 2;  //!< addresse
      uint8_t      : 2;  //!< EMTPY
    };
    uint8_t raw;
  } status_t;
  status_t _status;
```

Must get updated to

```
typedef union {
     struct {
       struct __attribute__((__packed__)) {
         bool por   : 1;  //!< status: power on reset
         bool crccfg : 1;  //!< status: crc
         bool dr    : 1;  //!< status: data ready
       };
       uint8_t      : 1;  //!< !addr[0]
       uint8_t addr : 2;  //!< addresse
      uint8_t      : 2;  //!< EMTPY
    };
    uint8_t raw;
  } status_t;
  status_t _status;
```

There's an issue on the ESP32S3 specifically where the bit assignment in the union doesn't work properly unless explicitly states.

You can also optionally add `_status = read(&adcdata);` outside of the 

```
while (!_status.dr) {
       _status = read(&adcdata);
}
```

in analogRead; however, I'm not sure if this is necessary or not.

Best,

nerdyscout, 2022-12-09:
could you please specify what you mean by:
`Also dude to the layout error in swap the pin definition of MISO & MOSI in the class Constructor .`
where did I swap something?

FelixWT, 2022-12-12:
> could you please specify what you mean by: `Also dude to the layout error in swap the pin definition of MISO & MOSI in the class Constructor .` where did I swap something?

This is the problem on my end. It is dude to the wrong PCB layout of my project.
thank you for your replay

FelixWT, 2022-12-12:
[Mirageofmage](https://github.com/Mirageofmage)
Thank you for your quick reply , I've tried what you suggest but it seems not working for me. It still get stuck in the loop. 

By the way, I should  mention this in my post that I'm using MCP3562R: dual channel variant.

This is my circuit:
[image]

I'm trying to read ADC readings from Device 1. So I'm assuming I should read both MCP_CH0 and MCP_CH1 manually to calculate voltage.

nerdyscout, 2022-12-12:
I do not have a  `_spi->begin(); ` which takes any parameters, therefore no :)

Please check the examples, there I use a setup with a custom SPI
```
SPIClass mySPI(&sercom1, 12, 13, 11, SPI_PAD_0_SCK_1, SERCOM_RX_PAD_3);
MCP3561 mcp(8, 7, 10, &mySPI, 11, 12, 13);
...
  _spi        = theSPI;
...
 _spi->begin(); 
```

As you can the the custom SPI `mySPI` is passed through the constructor into the privat member `_spi` which therefore already got the correct pin definitions.

FelixWT, 2022-12-13:
> There is no `begin() ` in SPIClass which takes any parameters, therefore no :)

Well , it seem that Arduino core didn't provide it but in ESP32 core it does. Check out the following link for SPI library in esp32 core
(http://github.com/espressif/arduino-esp32/blob/9275dbff942bb2ccce62632607500ad27dd0513a/libraries/SPI/src/SPI.h#L63)
And yes I did try out your code as well. But again it seems to be Arduino and ESP32 core diffrence I can't get your configurations to work.

> int32_t adcdata = mcp.analogRead(MCP_DIFFA);

Ah, you did provide the settings I want. This is great it feels weird to modifiy your library.
Thanks for this amazing library!

closed 2022-12-13 by FelixWT

FelixWT, 2022-12-14:
This isn't relate to this topic , but can I read the register of the MCP3561 through SPI?