nerdyscout/Arduino_MCP3x6x_Library #8: How is the “_status.dr” logic changed?

arthanth, 2022-12-27:
How is the “_status.dr” logic changed?
(In the IRQ_handler() does not change the “_status.dr”)
Thanks

Mirageofmage, 2022-12-27:
_status.dr is updated through a union based on the result of a function call. See the source which defines the structure below, specifically line 113.

https://github.com/nerdyscout/Arduino_MCP3x6x_Library/blob/c9de707794ecda62178ce5cdd86d50e9d86584eb/MCP3x6x.h#L108-L121

`IRQ_handler` calls the `read` function, which calls the `transfer` function, which in turn sets the value of `_status.dr` based on the result of the SPI transfer

https://github.com/nerdyscout/Arduino_MCP3x6x_Library/blob/6cce10135956cceaa817b629f27d1a70c646e8a2/lib/MCP3x6x/MCP3x6x.cpp#L80-L90

Is _status.dr not updating? If it isn't try using the [`develop`](https://github.com/nerdyscout/Arduino_MCP3x6x_Library/tree/develop) branch since commits to that branch may solve your issue.

arthanth, 2022-12-27:
Hello Mirageofmage,
Thank you very much for fast reply.
In the "_transfer" function sets the "_status.raw" value 
which part of the "_transfer" function changes the “_status.dr” logic?
Thanks.

nerdyscout, 2022-12-27:
`_status.dr` is just one bit within `_status.` the return value of `_transfer` is bit aligned to this status, just like Mirageofmage pointed out with the `status_ union`

nerdyscout, 2022-12-27:
@arthanth is this some real issue or are you just trying to understand the code? if first please specify your problem.

arthanth, 2022-12-27:
I use MCP3564 (ADC 9 Click) with ESP32 (Wemos D1 R32 pinOut-R32). Voltage incorect value it show 0 and 2.39 in mux mode. I try to fix follow the previous issue but it does not work now. So I am trying to understand the code and debuging.

FelixWT, 2022-12-29:
> Voltage incorect value it show 0 and 2.39 in mux mod

You should check the configuration 0 register of your MPC3564. The default Voltage Reference is set to 2.4V.

[image]

nerdyscout, 2022-12-29:
a few more thoughts:
- check if you got a version mcp3x6x**r** or without **r**, the once without r do not have VREF_SEL. did you connect anything to VREF+-?
- it helps a lot to have a logic analyzer / oscilloscope running in parallel to your SPI
  - I just checked if the code builds for ESP32, but never actually connected one. you might need to change   `_spi->begin();` adding the actual pins. Also just yesterday figured out there is a   `_spi->transfer32();` for at least some ESP32 variants... should not matter to you, but might help to speed up communication.
- `mcp.begin()` does set some defaults, which are not necessarily the defaults according the datasheet. all settings done by you (select channel, clk_sek, vref, gain, timing, ...) should be done after the `begin()`

nerdyscout, 2023-01-04:
I am closing this as no actual question/issue seems open. please feel free to come back if you run into troubles again.

closed 2023-01-04 by nerdyscout