nerdyscout/Arduino_MCP3x6x_Library #9: Difficulty with compiling for RP2040 using earlephilhower core

hallvardkristiansen, 2023-04-01:
Hi!
I'm having some trouble compiling my code after adding this library.
Most likely I'm doing something dumb, but would appreciate any input.

I include and initialise it like this:
```
#include "lib/MCP3x6x/MCP3x6x.h"
MCP3464 ext_mcp_adc(ADC_CS_PIN, &SPI, ADC_MOSI_PIN, ADC_MISO_PIN, ADC_SCK_PIN);
```

Then I get the following errors:
```
Linking .pio/build/PicoADK/firmware.elf
/Users/hallwrk/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: .pio/build/PicoADK/src/main.cpp.o: in function `_GLOBAL__sub_I_tft':
main.cpp:(.text.startup._GLOBAL__sub_I_tft+0x42): undefined reference to `_ZN7MCP3x6xC2EhhthPN7arduino11HardwareSPIEhhh'
/Users/hallwrk/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: .pio/build/PicoADK/src/main.cpp.o: in function `loop':
main.cpp:(.text.loop+0x98): undefined reference to `_ZN7MCP3x6x20analogReadContinuousENS_3MuxE'
```
(and so on)

I have some other peripherals on the two SPI channels which are working fine, an OLED and a DAC.

Mirageofmage, 2023-04-01:
Are you able to share additional code? It looks like the "loop" function is missing

Mirageofmage, 2023-04-01:
Are you including the Arduino library? (I don't know if there's RP2940 support).  It looks like this library is calling Arduino functions that don't exist

hallvardkristiansen, 2023-04-01:
Sure thing!
Everything not related to the ADC is working as it should.

Explicitly including Arduino.h doesn't make a difference.
Then again, the arduino stuff should be there from the platform.io config file I would think?

The adc.h file I am including is for an unused onboard ADC on the picoadk board.

```
[env:PicoADK]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
framework = arduino
board_build.core = earlephilhower
lib_deps = 
	bodmer/TFT_eSPI@^2.5.21
	nerdyscout/MCP3x6x@^0.0.1
build_unflags = -fno-exceptions
build_flags = blahblah
```

Here is the full main.cpp:

```
#include <TFT_eSPI.h>
#include <SPI.h>
#include <Wire.h>

TFT_eSPI tft = TFT_eSPI();  

SPISettings spiSettingsBoardADC(16000000, MSBFIRST, SPI_MODE2);
SPISettings spiSettingsExtDAC(20000000, MSBFIRST, SPI_MODE0);
SPISettings spiSettingsExtADC(20000000, MSBFIRST, SPI_MODE0);

#include "lib/MCP3x6x/MCP3x6x.h"
MCP3464 ext_mcp_adc(ADC_CS_PIN, &SPI, ADC_MOSI_PIN, ADC_MISO_PIN, ADC_SCK_PIN);

#include <adc.h>
#include <dac.h>

using namespace std;

#include <globals.h>
#include <functions.h>

void setup() {
  Serial.begin();
  Wire.begin();
  SPI.begin();
  SPI1.begin();
  delay(100);
  configurePins();
  initialize_gpio_mcp();
  initialize_ext_dac();
  tft.init();
  tft.setRotation(2);
  tft.fillScreen(TFT_BLACK);

  ext_mcp_adc.begin();
  ext_mcp_adc.enableScanChannel(MCP_CH0);
  ext_mcp_adc.enableScanChannel(MCP_CH1);
  ext_mcp_adc.enableScanChannel(MCP_CH2);
  ext_mcp_adc.enableScanChannel(MCP_CH3);
  ext_mcp_adc.enableScanChannel(MCP_CH4);
  ext_mcp_adc.enableScanChannel(MCP_CH5);
  ext_mcp_adc.enableScanChannel(MCP_CH6);
  ext_mcp_adc.enableScanChannel(MCP_CH7);
  ext_mcp_adc.startContinuous();
}

long lastDraw1 = guiStatus0.centiHertz;
long lastDraw2 = 0;
void loop() {
  millisTime = millis() % 1000; // Reset to zero every second
  microsTime = micros() % 1000000;

  readEncoders();
  if (millisTime - displayTime >= DISPLAY_HZ_MS) {
    readButtons();
    displayTime = millisTime;
    char buffer [33];
    tft.setTextSize(1);
    tft.setTextColor(TFT_BLACK, TFT_BLACK);
    tft.drawString(itoa(lastDraw1, buffer, 10), 0, 0, 2);
    tft.setTextColor(TFT_GREEN, TFT_BLACK);
    tft.drawString(itoa(guiStatus0.centiHertz, buffer, 10), 0, 0, 2);
    int32_t adcdata0 = ext_mcp_adc.analogReadContinuous(MCP_CH0);
    tft.setTextColor(TFT_BLACK, TFT_BLACK);
    tft.drawString(itoa(lastDraw2, buffer, 10), 0, 16, 2);
    tft.setTextColor(TFT_GREEN, TFT_BLACK);
    tft.drawString(itoa(adcdata0, buffer, 10), 0, 16, 2);
    lastDraw1 = guiStatus0.centiHertz;
    lastDraw2 = adcdata0;
  }
}
```

Mirageofmage, 2023-04-01:
Are your includes in order? As far as I know, the SPI library should be included before TFT

hallvardkristiansen, 2023-04-01:
Ah, well, yeah I guess it should be, but it is also included from within the tft_espi library, so it doesn't make any difference. I swapped them around now and that doesn't change anything.

Mirageofmage, 2023-04-01:
Are you building with GCC or an IDE?

hallvardkristiansen, 2023-04-01:
I am building with platform.io in vscode, it is using GCC?

Mirageofmage, 2023-04-01:
I'm not familiar with how platformio works in vscode, but how did you install the library/where did the 0.0.1 version number come from?

hallvardkristiansen, 2023-04-01:
I have tried installing it via the platform.io library manager and manually copying the code from this repo, which gave me the same result. V0.0.1 is what is given in the platform.io library manager version.

I believe it has something to do with how the core is written for Arduino. May have to ask Earle Philhower. Will also test the examples, but I expect the same result since there isn't really that much difference. Worth testing anyway :)

hallvardkristiansen, 2023-04-01:
Ah. The installation of the library via the platform.io library manager seems to be bad. I redid it entirely manually and uninstalled the one in the library manager and now it works...

closed 2023-04-01 by hallvardkristiansen