nerdyscout/Arduino_MCP3x6x_Library #10: Some questions about the SPI

FelixWT, 2023-04-07:
Recently I run in to some issue that got me good : that is the timming of CS pin pulldown
https://github.com/nerdyscout/Arduino_MCP3x6x_Library/blob/6cce10135956cceaa817b629f27d1a70c646e8a2/lib/MCP3x6x/MCP3x6x.cpp#L80-L90
As above in your code CS pin is  pulldowned after `beginTransaction` function been called , and it works! But when i'm using another adc (Texas Instrument's ads8691) it have to pulldown before it 
So my question is how do I know when do I need to place it before or after  `beginTransaction` function call ?

Another question: does the `noInterrupts(); ` effect on the Interrupts that  was pinned to another esp32 core?

Mirageofmage, 2023-04-07:
The CS pin is typically pulled low after `beginTransaction`. See the Arduino `SPITransaction` [documentation](https://docs.arduino.cc/tutorials/communication/SPITransaction).

The `interrupts()` and `noInterrupts()` can be removed.

FelixWT, 2023-04-08:
> The CS pin is typically pulled low after `beginTransaction`

This is what I thought at first , But for ads8691 it's not  .I assume I have to trail error when ever it work abnormally?

> Also, are you trying to use this library with the ads8691 ADC?

No I use this amazing library as the base structure for that ADC

nerdyscout, 2023-04-08:
hi @FelixWT,
I wrote this lib primarily with the SAMD21 in mind I am using it with but tried to keep it open to many other controllers. 
So I never tried any ESP by myself, hard to say what a multi core mcu does.

Setting up SPI Settings first, than pulling CS low seems the logical way to me. did you double check settings according to "Table 7-8. SPI Protocols for Configuring the Device"

@Mirageofmage I would be careful about removing noInterrupts(). I had some bad effects doing so! But you could also run into trouble as I assume no other interrupts (USB, watchdog , wifi, ....) will be handled while running though this code

FelixWT, 2023-04-10:
> did you double check settings according to "7.5.4.1 Protocols for Configuring the Device"
yes I did, it only discribes the SPI Mode. I assume it's something to do with the recognition time of CS PIN change.

closed 2023-04-10 by FelixWT