nerdyscout/Arduino_MCP3x6x_Library #11: Strange issue with MCP3561

y3nd, 2023-07-04:
Hello,
I know the question is not directly related to the library, but I still want to give it a shot, since I am a bit stuck there.

I have a MCP3561 with a MAX4208 amplifier to measure the voltage at the output of a strain gauge. The MCP3561 is drived by an ESP32 using this library. I would like the system to work in continuous mode in MUX mode (differential).
Everything is working well (config settings, continuous mode, irq pin) but I have very strange steps showing in the output signal. It seems to be related to some specific ADC count values, and I can't find anything similar on the web too. After some digging I found out that sometimes, somehow, a group of samples has missing ADCDATA bytes, always the less significant ones. So data like 0x3Cxxxx becomes 0x3C0000 and it stays like this until a high/low enough sample appears and change the current value.

[image]
Like you can see on this picture (values are calibrated values for my use), there are somes value-based steps appearing at 11.27-21.68-31.04, on the ascent and the descent of the curve.

There is a small antialiasing filter consisting of two RC (1k/100n) filters in between, like in the schematics of the [weight scale example application](https://ww1.microchip.com/downloads/aemDocuments/documents/MSLD/ProductDocuments/BoardDesignFiles/ARD00906_SCH.PDF) provided by microchip

I have tried with a signal generator directly on the input of the antialiasing filter and it still the same.

If you have any clue or something I could check/test/do to know what's causing this issue...

nerdyscout, 2023-07-04:
hi @y3nd 
are you using main or develop? they diverged currently a lot. of course main should be stable to use, but I suggest develop for new projects as there are some breaking changes and you might run into trouble adopting afterwards... but honestly I can not guarantee develop is working currently.

I can not remember ever experiencing your issues... a dirty fix which came to my mind is to increase [OSR](https://github.com/nerdyscout/Arduino_MCP3x6x_Library/blob/228a2bd98141f9e9c312e403f15db77cb89edbf0/src/MCP3x6x.cpp#L378)

more likely it is an issue with [_getValue](https://github.com/nerdyscout/Arduino_MCP3x6x_Library/blob/6cce10135956cceaa817b629f27d1a70c646e8a2/lib/MCP3x6x/MCP3x6x.cpp#L248C18-L248C27) - which data format are you using? as far as I remember this function changed as well between those branches...

y3nd, 2023-07-04:
I forked the main branch in December, i'll try the develop branch next but I don't think it's a conversion issue from the lib, since printing the raw values from the SPI transfer are showing the same 00000 for LSB. I saw on the web issues with SAR, and AD has [an article](https://www.analog.com/en/analog-dialogue/articles/precision-successive-approximation-adcs.html) about "stuck codes" but it doesn't match with the "specific stuck codes" issue I have, and my board has practically no noise as the analog parts are quite decoupled with their own power-supply etc.. but it could be flawed somewhere.
It might be an issue related to impedance matching with our amplifier, or specific high frequency noise affecting the ADC I don't really understand.
I have contacted Microchip support too.

y3nd, 2023-09-14:
I circumvented the ADC spikes issue by using differential mode. No spikes or stuck codes now, but really high gain all the time, about x20 gain. Setting the gain just multiplies the always-present x20 gain (0.33x = 6.6x, 1x = 20x, 2x = 40x etc..) really weird. I've contacted Microchip and they have no idea of what's going on. It might be a counterfeit, but we bought it through JLCPCB...

closed 2026-11-15 by nerdyscout