nerdyscout/Arduino_MCP3x6x_Library #12: How do I add license in my source code ?

FelixWT, 2023-09-25:
Hi,

I noticed that you have two licenses: CC0 and MIT. However, they seem pretty identical to me. Is it legal to include just one of the licenses?

Also, this is my first time adding a license. I don't know what the typical approach is for doing this. Should I add the full text of MIT/CC0 in the specific library, or should I create another folder to hold all the licenses?

nerdyscout, 2023-09-25:
hi @FelixWT 
different licenses apply to different files. which one apply to which can be found in /.reuse/dep5
personally I don't have a lot experience with all this legal stuff... for me it is fine if you put the MIT somewhere.

FelixWT, 2023-09-28:
Ok, thanks for the info

closed 2023-09-28 by FelixWT