nerdyscout/Arduino_MCP3x6x_Library #13: ESP32 and MCP3462 not working in Mux Mode

LeKrakenTheCode, 2024-01-24:
Keep receiving a value of zero from analogRead. Moving this into its own issue from: https://github.com/nerdyscout/Arduino_MCP3x6x_Library/pull/5#issuecomment-1870591920

LeKrakenTheCode, 2024-01-24:
> What's the speed of your SPI bus? I remember having some issues if the speed was over 500k. I'm currently running w/ 200k
I was running at 10MHz. I lowered to 200KHz and still no change.

Mirageofmage, 2024-01-24:
If you have access to a scope, can you check the activity on MISO?

LeKrakenTheCode, 2024-01-29:
Sorry for the delay. Was feeling rather unwell this weekend. Working on it now.

nerdyscout, 2024-02-03:
MISO and MOSI seem to show reasonable signals. Seems like the second picture is showing a 1Byte command followed by another 2Byte command... I did not try to figure out which are these. But it seems save to say `MCP3x6x::read(Adcdata *data)` is never executed at this would result in a 3-5Byte command. At least this is not shown on any of the pictures.

I still suspect the issue in this little snippet:
````
    settings.mux = ch;
    _status      = write(settings.mux);
    _status      = conversion();
    while (!_status.dr) {
      _status = read(&adcdata);
    }
````

please put an `Serial.print(_status.dr)` inside the loop and right in front of it, it looks like this is currently not executed. This bit should be set when conversion started and reset when valid data available. As this bit is part of the status byte endianness is important, but also timing of spi speed vs conversion speed might cause it.

LeKrakenTheCode, 2024-02-05:
[image]
Here's the change and here is the output:
[image]
only one appears to be running.

Mirageofmage, 2024-02-05:
> ![image](https://private-user-images.githubusercontent.com/47678961/302366122-a40b8e75-beb8-4ac2-9da2-ac087e920a26.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

Remove the while loop, so it should directly read the adcdata reference instead of waiting for _status.dr

Mirageofmage, 2024-02-05:
> ![image](https://private-user-images.githubusercontent.com/47678961/302366122-a40b8e75-beb8-4ac2-9da2-ac087e920a26.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

I've had issues with the while loop, in my implementation I just directly read instead of waiting for the data ready bit. I have to be careful with my sample rate or it returns the same value for each of the channels.

LeKrakenTheCode, 2024-02-05:
[image]

No change to the output.

Mirageofmage, 2024-02-05:
[image]

Mirageofmage, 2024-02-05:
[image]

LeKrakenTheCode, 2024-02-05:
still no change to the output:

[image]

Mirageofmage, 2024-02-05:
@nerdyscout 
[image]

Question on why the int32 is truncated to 25 bits in adcdata? The ADC can return either left aligned or right aligned 32 bit values. The assignment in the result struct retains all 32 bits

nerdyscout, 2024-02-05:
>
> Question on why the int32 is truncated to 25 bits in adcdata? The ADC can return either left aligned or right aligned 32 bit values. The assignment in the result struct retains all 32 bits

yes, the dataformat is choosen within the begin() funtion

Mirageofmage, 2024-02-05:
> > Question on why the int32 is truncated to 25 bits in adcdata? The ADC can return either left aligned or right aligned 32 bit values. The assignment in the result struct retains all 32 bits
> 
> yes, the dataformat is choosen within the begin() funtion

I see that, but my point is that if somebody chooses a left aligned output, then the 7 leftmost bits get masked out? Or am I understanding it wrong?

nerdyscout, 2024-02-05:
I assume you are right. 
My intention was to make this library for getting easy started as the component is really complex. This is why the begin functions sets a few default parameters... they might be not good for every case, but if someone chooses to set up their own config than I would suspect it is on him self to handle this.

LeKrakenTheCode, 2024-02-12:
Any other idea's I can try?

Mirageofmage, 2024-02-12:
> Any other idea's I can try?

Maybe try returning `adcdata.value` directly at the end of analogRead?

So
`return adcdata.value;`
instead of
`return result.raw[(...`

Otherwise I don't really have a clue/it has to be in hardware? I've been testing the same code revisions on my hardware and it's been working (though I'm not using the R revision).

I remember that you put the pins in the constructor of whichever MCP3x6x class you used, I instead opted to redefine the consts in the default constructor.

So my object declaration is just
`MCP3561 mcp;`

with #defines for MOSI, MISO, SS, SCK, and IRQ (if required). The default SPI interface through Arduino is selected.

Mirageofmage, 2024-02-12:
I actually think you _**HAVE**_ to use #define to select pins instead of using the constructor, it doesn't look like parameters are properly passed through the constructor to the SPI object

The parameters required are 

```
#define MOSI 
#define MISO
#define SS
#define SCK
```

nerdyscout, 2024-02-12:
> I actually think you _**HAVE**_ to use #define to select pins instead of using the constructor, it doesn't look like parameters are properly passed through the constructor to the SPI object
> 
> The parameters required are
> 
> ```
> #define MOSI 
> #define MISO
> #define SS
> #define SCK
> ```

as his logic analyzer pictures have proven some SPI communication I don't see why this should be the case.

Mirageofmage, 2024-02-12:
If he's hooked up to the default pins then this isn't necessary, I'm just assuming not since he passed in parameters to the constructor.

nerdyscout, 2024-02-12:
@LeKrakenTheCode  please publish your code. either setup a new repository as minimal example which reproduces your issue or the original code.

LeKrakenTheCode, 2024-02-12:
It has been published here:

https://github.com/McMillanFlowProducts/McMillanDevBoard/

LeKrakenTheCode, 2024-02-12:
I had to make a change to the line 69 in the cpp:
 _spi->begin(_pinCLK, _pinMISO, _pinMOSI, _pinCS);
To get it to work with the ESP32 in the first place.

nerdyscout, 2024-02-20:
seems like it could be fixed by something like this, will work on a better fix later.
````
MCP3x6x::status_t MCP3x6x::_transfer(uint8_t *data, uint8_t addr, size_t size) {
  _spi->beginTransaction(SPISettings(MCP3x6x_SPI_SPEED, MCP3x6x_SPI_ORDER, MCP3x6x_SPI_MODE));
  noInterrupts();
  digitalWrite(_pinCS, LOW);
  _status.raw = _spi->transfer(addr);
#ifdef ARDUINO_ARCH_ESP32
  _spi->writeBytes(data, size);
#else
  _spi->transfer(data, size);
#endif
  digitalWrite(_pinCS, HIGH);
  interrupts();
  _spi->endTransaction();

  return _status;
}
````

McMillanFlowProducts, 2024-02-21:
Just changed the _transfer function, still getting a reply of zero.

closed 2026-11-15 by nerdyscout