nerdyscout/Arduino_MCP3x6x_Library #17: chip fails to report anything from analogRead

KrisKasprzak, 2025-01-31:
Howdy,

I'm having issues getting a reading from my MCP3564 chip. I have tried 2 chips no luck.
If I run the example analogRead_Scan.ino (and comment the _status.dr at line 325

Serial.print("voltage0: "); returns 2.3999998569. In fact all voltages return this value

Since others seem to get this library working, I'm guessing it's on my side, hopefully you can help.

My Chip
MCP3564RT-E/ST
https://www.digikey.com/en/products/detail/microchip-technology/MCP3564RT-E-ST/12807301
chip soldered to a test PCB board with no other chips/components

My hardware
MCU Teensy 3.2
IDE Arduino 2.1.0
Windows 10 OS

Chip wiring
chip		Teensy
AVdd 		3v3
Agnd 		gnd
REFIN(-) 	N/C but no change if connected to Teensy GND
REF(+) 		N/C but no change if connected to Teensy GND
CHO 		(connected to Teensy GND or 3v3 in an attempt to get a reading)
CH1		N/C
CH2		N/C
CH3		N/C
CH4		N/C
CH5		N/C
DVdd 		3v3
Dgnd 		gnd
MCLK		N/C
IRQ		N/C
SDO 		pin 11
SDI		pin 12
SCK		Pin 13
cs		Pin 10
CH7		N/C
CH6		N/C

If if I run the test program analogRead_Scan I never get anything printed to the serial monitor

mcp.begin() does returns 1  and if I add code  
  Serial.print("getReference ");
  Serial.println(mcp.getReference(), 7);

Serial shows
getReference  = 2.4000001

in the method analogRead(mux_t ch), _status.dr after  _status = read(&adcdata);

always returns 0

If I comment out while (!_status.dr) {... around line 325

the code continues and Serial.print("voltage0: "); returns 2.3999998569 (looks like the internal reference voltage.

questions
1) is my wiring correct.
2) thoughts on what the problem could be?

Mirageofmage, 2025-01-31:
Since you don't have an external clock connected, make sure you set the Config0 register to select the internal clock.

[image]

Adding the line, `mcp.setClockSelection(clk_sel::INTERN);` should solve the issue but I believe this should already be done in the initialization function, is this right @nerdyscout? Otherwise, make sure the ADC is using the internal reference by calling `mcp.setReference(0);`

Can you attach a schematic to check connections?

KrisKasprzak, 2025-01-31:
@Mirageofmage 

thanks for the reply

adding
mcp.setReference(0);
changes nothing

adding
mcp.setClockSelection(clk_sel::INTERN);
fails to compile (error: 'clk_sel' has not been declared)

Apparently I can't add an image but wiring
chip Teensy
AVdd 3v3
Agnd gnd
REFIN(-) N/C but no change if connected to Teensy GND
REF(+) N/C but no change if connected to Teensy GND
CHO (connected to Teensy GND or 3v3 in an attempt to get a reading)
CH1 N/C
CH2 N/C
CH3 N/C
CH4 N/C
CH5 N/C
DVdd 3v3
Dgnd gnd
MCLK N/C
IRQ 100K pullup
SDO pin 11
SDI pin 12
SCK Pin 13
cs Pin 10
CH7 N/C
CH6 N/C

KrisKasprzak, 2025-01-31:
PS... thanks a million for jumping in.

KrisKasprzak, 2025-01-31:
[image]

KrisKasprzak, 2025-01-31:
If I unwire everything, meaning the chip is not connected to anything
 and I run this code

#include "MCP3x6x.h"

MCP3564 mcp;

void setup() {
  Serial.begin(115200);
  while (!Serial)
    ;
  Serial.println(__FILE__);

  bool test = mcp.begin();
  Serial.print("test: ");
  Serial.println(test);
}

// the loop routine runs over and over again forever:
void loop() {
}

  Serial prints 1.

Not sure how SPI can pass with nothing connected.

KrisKasprzak, 2025-01-31:
I have included decoupling caps and the series resistors per the data sheet (not shown in the above schematic). I looked at the low level spi call in the library where spi transfers are done and _spi->transfer returns 255

I really have no idea how to proceed.

MCP3x6x::status_t MCP3x6x::_transfer(uint8_t *data, uint8_t addr, size_t size) {

  _spi->beginTransaction(SPISettings(MCP3x6x_SPI_SPEED, MCP3x6x_SPI_ORDER, MCP3x6x_SPI_MODE));

  noInterrupts();

  digitalWrite(_pinCS, LOW);

  _status.raw = _spi->transfer(addr);

////////////////////////////////////////////////////////////////////////////////
// _status.raw returns 255
// _pinCS is correct (pin 10 in my case)
////////////////////////////////////////////////////////////////////////////////
 
  _spi->transfer(data, size);

  digitalWrite(_pinCS, HIGH);

  interrupts();

  _spi->endTransaction();

  return _status;

}

nerdyscout, 2025-02-01:
hi KrisKasprzak,
I will try to comment a few things which might help, but so far I did understand there are several things broken and I currently dont have the solution for everything.

- which commit of this lib are your currently running?
`main` is kind of old and will have some breaking changes once `develop` is merged. The best I currently would suggest is `dc400ee`
newer `develop` code do have a outdated examples and readme.

- Examples with `Scan` in the name MUST connect a irq pin! 
- Mux Examples never really worked because of an endless loop here `while (!_status.dr)`

- readings of ~2,4V are kind of a good sign as it means internal reverence has been configured. but is also means REF+ pin should be left floating (or just add a cap to GND)

- this old commit does set clock within `mcp.begin()` function, somewhen later this is removed and therefore has to be done manually.

- _status.raw of 255 is something totally new to me, dont know why. As this status contains the device address within the first 3 bit it should never reach this value.

as this project uses a teensy as well it might help you better than I can currently:
https://github.com/edmundsj/MCP3561DevBoard/tree/master/mcp3561_arduino

KrisKasprzak, 2025-02-01:
Thanks, I tried it an "something" worked one time. I wish I had a working wiring diagram.  Regardless this has to be the most finicky chip I've ever tried using....

Thanks anyway.

closed 2025-02-01 by KrisKasprzak